autoboat-vt/OLD-sailbot-21 #39: Port existing Arduino code to Teensy

evallen, 2020-10-19:
We need to take our existing Arduino code and move it to the Teensy. 

Please use this issue to discuss any progress or problems you encounter during this process.
This can include weird pin assignment changes, compilation problems, or more. 

This issue will be closed when all of our Arduino code compiles and operates correctly on the Teensy board. 

_This issue will remain on the sailbot-21 repo even though our Arduino code isn't necessarily on here for simplicity._

ahellerjones, 2020-10-30:
Played around with the 4.0 tonight; remarkably easy to get up on running on linux, which is actually commonly a huge pain, at least it can be for first party Arduinos. I downloaded the [teensyduino](https://aur.archlinux.org/packages/teensyduino/) package from the aur, no idea what it actually does, some strange GUI for uploading Intel HEX files (??) 

After that, followed the instructions from the [PJRC](https://www.pjrc.com/teensy/td_download.html) (manufacturer of the Teensy's) and it worked out of the box. You can even change the clock speed (OC to 1GHz??), change your upload speed / compactness, keyboard output, USB type etc. Maybe I just haven't fiddled around with the Arduino IDE in a while but it's a welcome sight to see expanded use ability. Played around with the Teensy's included examples, customized a Serial one to report back anything intercepted on the Serial Monitor and got everything back... things work! Didn't need any extra `#includes` that I was worried about. 

Next steps in a semi particular order: 
1) Figure what Arduino code we need to upload.
2) Upload said code 
3) Begin testing said code, expanding said code, and making it actually good. Would also to be nice to nail down what motors are going to be used. 
4) Get some gosh darn pin headers and solder them on. If anyone has extras laying around so that I don't have to actually get my warelab badge that'd be cool.

evallen, 2020-10-30:
Wow! That's really great to hear. I did not expect it to work so cleanly out of the box. 

Do you know if the pin assignments seem relatively similar?

ahellerjones, 2020-10-30:
Well I don't think we have anything really in concrete right now for pin assignments, or anything that can't easily be changed. Reassigning pins should be relatively straight forward. I can begin working on a simple over view schematic for review/

ahellerjones, 2020-11-24:
Starting from the ground up, no reassigning pins just tossing the Arduino code for controls last semester, it was all garbage (or I can't find it tee hee) so just starting fresh

evallen, 2021-02-25:
Notes from today - may want to implement some sort of watchdog timer or other means to reset the system if the Maestro blocks program execution, as happened today.

ahellerjones, 2021-02-26:
Looked into it this morning, it's actually fairly straight forward. 

https://circuits4you.com/2018/01/24/tutorial-on-arduino-watchdog-timer-setup/

evallen, 2021-02-26:
That's really convenient! Can you describe which parts of the Maestro code can block overall execution?

ahellerjones, 2021-02-26:
What I noticed yesterday is if there's a communication failure between the Maestro and the Arduino, the arduino will halt until it receives feedback. This could be seen when a test would halt while calling `servo.getCurrentDegrees()` while the corresponding Rx pins weren't connected. I think the safest route would be to but halting checks on all methods that interface with the Maestro, but for now we can just use a global check for the whole system.

evallen, 2021-02-26:
Hmm - that does surprise me. I wonder if there's a way to do it asynchronously?
A global watchdog seems good for now, though.