autoboat-vt/OLD-sailbot-21 #41: Develop object-to-waypoint conversion methods

evallen, 2020-10-19:
Our current autonomy code defines various _objectives_ for the boat in `src/autonomy/objectives.py` (_note: currently, the most up-to-date autonomy code is on the `autonomy_integration_tests` branch_). 

We need to develop methods that turn these objectives into series of waypoints (i.e., if the boat's objective is `ROUND_BUOYS_CCW_LOOPING`, what are the intermediate waypoints it should travel to?).

These are not the same as tack points, which are short-term points used to navigate upwind. 

We have implemented these methods for two objectives so far, found in `src/autonomy/nav/captain.py`'s `add_marks_for_objective` method. 

Please check off each objective as the waypoint conversion method is implemented (for objective descriptions, see `src/autonomy/objectives.py`). 
- [ ] ENTER_STARTING_GATE
- [ ] ROUND_BUOYS_CCW
- [ ] ROUND_BUOYS_CCW_LOOPING
- [ ] ENTER_SK_BOX
- [ ] STAY_IN_BOX
- [ ] LEAVE_BOX
- [ ] ENTER_SEARCH_AREA
- [ ] START_SEARCH_PATTERN
- [ ] TOUCH_ONE_BUOY

_This issue will be closed when we have developed objective-to-waypoint conversion methods for each objective listed above._

satwik343, 2020-10-25:
For the STAY IN BOX method, we can detect the position of the 4 buoys and also the angle between two adjacent buoys. 

When taking two buoys, we find the angle between the "line" from the boat to one buoy and the "line" from the boat to another adjacent buoy. *It would be best of the two buoys used were the ones that were closest in distance to the boat.* If the boat is moving toward the edges of the box(almost going out of the box), the angle between these two "lines" approaches 180 degrees. We can continuously check this angle, and if the angle is approaching 180, the boat can move in the opposite direction and toward the center of the box.

evallen, 2020-10-26:
This is actually a pretty clever algorithm. I think it would be cool to merge this with some of our GPS data - if we know the positions of the buoys, we can try to ensure we stay near the center coordinates of their box. This would be a great way to back that up in case the GPS is unreliable.

evallen, 2021-02-15:
Not planned for this year.

closed 2021-02-15 by evallen