autoboat-vt/OLD-sailbot-21 #42: Research / plan a more accurate boat simulator

evallen, 2020-10-19:
We need a simulator for our boat that is as realistic as possible to test our various autonomy and control programs because our opportunities to test the boat in real life will likely be limited. 

Our current simulations (found in `integration_tests/` of the `autonomy_integration_tests` branch) are great starting points, but they do not take into account more advanced factors such as wind direction, current control surface angles, and more. Implementing a system that would simulate the boat's behavior in the water based on these factors would allow us to create a much more advanced autonomy and controls system. 

Having one simulator would also allow us to easily create integration tests that place our boat into various scenarios.

In this issue, I would like for us to develop a plan for this simulator:
* **What kinds of factors should affect the simulation (wind direction, control surface angles, water drag, etc.)?**
We can always get more detailed, and at a certain point things may become more difficult to implement than they are worth. We should decide which factors we want to focus on. 
* **How would we implement those factors?** For example, how can we simulate the effect of wind on the boat? We probably should discuss this with Sail team or NavArch. Please place online sources in this issue if you find any that are helpful. 

Once we have decided on these things, I will open an issue to actually implement the simulator.

evallen, 2020-11-05:
There's an interesting paper on nearly exactly this [here](http://ceur-ws.org/Vol-2331/paper3.pdf).

Gcambridge, 2020-12-18:
Here's what I got from the first few pages of the paper:

**Negligible Factors and Assumptions:**
- The change in wind speed and angle across the sail - use a constant speed and angle for full area of the sail. 
- The change in currents across submerged elements such as keel and rudder - use a similar approach to wind speed variation and assume constant speed and angle for current speeds. 
- The change in flow of both current and wind across lateral areas of the ship - use constant speed and angle for wind and current across the area of elements such as the hull and any surface besides the primary control surfaces. 
- Interference between multiple sails such as a reduction in flow from a jib to mainsail - characterize sails individually using "thin airfoil theory" - _research this or discuss with sail team_
- Assume wavelength of surface waves is relatively large to the boat length

**Relevant forces and components:**
- Forces on Keel, Rudder, and Sail
- Buoyancy and forces from waves - forces such as waves altering heading, creating resistance, or altering pitch <- _How relevant is this? Are conditions at competitions typically wavy or calm. Does this functionality already exist? Is it complicated to include this functionality in both the simulator and autonomy software?_ 
- Damping forces - _I think this might be referring to ballast, further research needed_

I can also look more into the characterization of boat hulls to see what relevant factors regarding hull design should be included in the simulator. This is just what I pulled from the assumptions and description of model components.

evallen, 2020-12-23:
Thanks @Gcambridge. This is a good list of things to strive for in our simulation in the long term. Right now, I want to focus on getting a very basic standalone simulation working. 

@MichaelKattwinkel is currently working on this - can you bring @Gcambridge up to date on what you've got so far?

MichaelKattwinkel, 2020-12-23:
@Gcambridge I am currently working on getting used to pygame (the library I'm going to use for the simulator).
My next step is to be able to dynamically plot the boats movement and rotation

evallen, 2021-02-15:
No longer planned for this semester.

closed 2021-02-15 by evallen