autoboat-vt/OLD-sailbot-21 #44: Establish basic LiDAR access over Teensy

evallen, 2020-10-19:
Get basic communication going between the LiDAR and the Teensy. 

We should be able to send basic messages back and forth between the LiDAR and the Teensy. Once we can do that, we can open another issue to fully integrate it into our sensors system.

ahellerjones, 2020-11-24:
This issue might be able to be moved to the Arduino repo, LiDAR is still very much on the back burner for now :( but hopefully soon!

evallen, 2020-11-24:
Yeah, soon hopefully!
I agree that we could move this issue to the Arduino repo, but I wanted to keep the issue here just so we had a central place to gather all of the project's issues (even if the code technically isn't on this repo) and we don't forget about some on an alternate repository.

evallen, 2020-11-24:
Actually, I see that you opened a couple of issues already on the Arduino repo, and I do like how that's seems to work. I'll move this one.

closed 2020-11-24 by evallen