autoboat-vt/OLD-sailbot-21 #45: Investigate potential improvements / replacements for object detection system

evallen, 2020-10-19:
We need to improve our object detection system to more reliably detect buoys and boats in the water. It currently can only detect buoys, and its performance wavers under different lighting conditions. It has only been tested on images of the buoy taken at the Drillfield & Duck Pond. 

It currently uses a color-matching system to detect buoys (the vast majority of this color detection system involves a single-time analysis that lives on [vt-sailbot/sensor-eval's color_analysis branch](https://github.com/vt-sailbot/sensor-eval/tree/color_analysis) since that only needs to be done once and doesn't need to be on the boat; Step 5 and 6 are implemented here on this repo since the boat does need to do those).

Essentially:
1. We take lots of pictures with / without buoys. 
2. We label the pictures, creating "masks" for each image that contain either black or white pixels. The white pixels are the pixels of the buoy, and the black pixels are anything else. 
3. For each image, we create histograms of all the colors in the image. We create 6 histograms: 3 for the Hue, Saturation, and Value (HSV) respectively of every pixel marked as "buoy" in step 2, and 3 more for the HSV of every pixel marked as "not buoy". 
4. We save the _difference_ between the two histograms so that colors that are very present in buoy pixels but not in other pixels are scored highly.
5. When we actually analyze a picture to find a buoy, we score each pixel by looking up that pixel's HSV value in the saved difference histograms from step 4. We call this score the "relevance", and it produces a "relevance map" (it looks like a heat map with the "hot" or "white" regions being the ones that have the colors that we identified as a buoy). 
6. We use computer vision techniques to identify white blobs in the relevance map, and call those buoys.

This approach works okay, but doesn't seem very robust under weird lighting conditions. In addition, large changes in the picture environment (like moving from the green drillfield to the blue water, for example) would likely require a second color analysis, which takes a long time since we have to meticulously label each buoy pixel (a bounding box isn't enough - we have to get the buoy almost perfectly). It also only works on buoys whose color we already know. 

**Please investigate various ways we can either improve this system to get better performance or other alternative systems we can create to make our object detection system more accurate (and work on boats too!).** 

Potential priorities:
* Accuracy
* Ease of implementation
* Ease of data collection / labeling (do we need to get each individual pixel, or is a bounding box "good enough"?)
* Speed of execution on embedded devices (hopefully we can get an NVIDIA Jetson Nano, which would let us do some crazy ML stuff if we want!)
* Robustness to changes in lighting / environment (we have a test buoy, but it may not be an exact match to the ones they'll use at competition). 
* Extensibility for other objects - like other boats

Please place your discussion in this issue.

ibotismariah13, 2020-11-06:
This is the google doc I am using to put links to some interesting vision algorithms. I am also not sure what we're doing for vision processing so far, but I put some links to cameras and software that I've seen be easy to work with for vision processing. (https://docs.google.com/document/d/11iX6-iTGaUaOlp7wNf3l33OwOdrcgJ6UzBY1L1_3oCw/edit?usp=sharing )

evallen, 2021-02-15:
Not planned for this year.

closed 2021-02-15 by evallen