autoboat-vt/OLD-sailbot-21 #46: Investigate potential new boat computers

evallen, 2020-10-28:
Our current Beaglebone Black is a little slow for intensive computation. This might limit our ability to process sensor data with more advanced techniques (such as machine learning). 
Please investigate other boards we could potentially use to replace it. 

Goals for an ideal replacement board:
* Powerful enough to run standard image-based machine learning algorithms (such as [YOLO](https://github.com/ultralytics/yolov5)) at a reasonable frame rate (ideally > 6-8 frames per second if possible; low frame rates should be fine for basic operation, but higher frame rates would let us sweep the cameras faster and have more situational awareness). 
* Low-power enough to last the length of the endurance race (7 hours) - (this is a _huge stretch_ and it's okay if it doesn't actually last that long, since we can replace the batteries in the middle of the race for a small deduction - consider this a "nice-to-have" as long as the board lasts at least the length of the Beaglebone Black).
Determining the life of the board will involve analysis of our current power systems capabilities - documentation about that is in the shared drive under ECE > 2019-20 > Power Systems, but we can talk about that separately since it might be involved.
* Has enough / proper peripherals to drive Teensy and other devices (cameras, LiDAR, etc.). 
* < a few hundred dollars at most

Please feel free to edit / add to these goals if you feel like they should be changed for our application.

We have already been looking at the NVIDIA Jetson Nano, but we should investigate if there are any better alternatives.

ahellerjones, 2020-10-30:
First notes: YOLO relies on NVIDIA CUDA cores, parallel processing cores within their GPU's -- which explains why NVIDIA boards should most likely be prioritized... [In fact Nvidia boards are the only boards that will support running this YOLO software.  ](https://pjreddie.com/darknet/install/#cuda)

ahellerjones, 2020-10-30:
I think the best approach is to use nvidia first party Boards, (like the nano), as I believe this is going to have the largest amount of support / community behind it, and documentation. 

Options: 
[Jetson Nano](https://www.nvidia.com/en-us/autonomous-machines/embedded-systems/jetson-nano/) @ $100. 
[Jetson Xavier NX](https://www.nvidia.com/en-us/autonomous-machines/embedded-systems/jetson-xavier-nx/)

[Sparknotes from here:
](https://www.seeedstudio.com/blog/2020/06/04/nvidia-jetson-nano-and-jetson-xavier-nx-comparison-specifications-benchmarking-container-demos-and-custom-model-inference/) Xavier boasts USB 3.1 which will have a higher throughput to USB devices, albeit I don't even know what devices we have that'll be USB enabled. 
The Xavier has two M.2 slots, one supporting an NVME SSD, albeit an NVME would probably be overkill for our application. 
Xavier requires a 19V power supply -- this is probably out of our range. 
In the article, the author comments on a user trained model that was made, something about identifying dogs running "real time from a USB web camera", the Nano averaged 15 FPS, while the xavier could process 30fps. The Nano is within our tolerances, and probably all we would need. 

On the site above's 
Jetson AGX Xavier HA just kidding this thing is $700 i don't think we need anything the Xavier

evallen, 2020-10-30:
Yeah, that's about what I thought. YOLO I believe can run on a CPU without CUDA, but it is much, much slower. 

We aren't committed to YOLO or a CUDA-based system yet, but I think you bring up a very important point so we don't limit our analysis options.

evallen, 2020-11-05:
Going to go with the NVIDIA Jetson Nano.

closed 2020-11-05 by evallen